uc-vision/RovVision2 #52: At the end of the automated scan, the rov slides to the next set of lines.

codedraughtsman, 2023-07-18:
At the end of the scan, the rov comes up the last dropper line to the set target depth.
Then it slides across to the next  dropper line before finishing the automatic scan.

I consider this unneeded slide to be confusing as it makes me think it is starting another scan. 

Can we make it so the ROV only slides if it needs to scan the next dropper line.

orig74, 2023-07-18:
It's not a bug. it's designed to support consecutive scans. otherwise it will start the next scan at the last already scanned line.

codedraughtsman, 2023-07-31:
@orig74 that's understandable. 

My argument is that the most common use case is to only execute a single automated scan, then retrieve the ROV and head to the next scan site.

Personally, I would expect to manually have to move the ROV over by a line before starting a new scan.

My suggestion is we make this into a "slide at end" option that is set to false by default.

closed 2023-09-18 by TimKeesRensen