uc-vision/RovVision2 #54: toggling "mission start" after rov has completed an automated scan, does not start another scan

codedraughtsman, 2023-07-18:
Steps used to create this bug:
Start sim
set the max_iters to 1

drive up to line and start the scan.

after the scan is finished, the auto-scan state should be in "stabilise"
untick and retick the "mission start" checkbox.

Rov should start another scan, however it remains in the stabilise mode and doesn't move at all.

rearming the ROV doesn't fix this. Suspect that it would require restarting the rov to fix/clear it.

codedraughtsman, 2023-07-18:
Have tried and failed to reproduce this twice. Wondering if this could have been an issue with the ROV not stabilising enough for it to move in to the "going down" state in the second scan.

orig74, 2023-07-18:
yes,
you need to wait for it to stabilize.

closed 2023-07-18 by orig74