uc-vision/RovVision2 #57: Setting D405 camera to a high exposure crashes GUI

codedraughtsman, 2023-07-27:
At a field trip we crashed the GUI by setting the D405 camera's exposure to 30. 
after rebooting the GUI, we continued with a exposure of 20, which seemed to be fine.

Can we limit the maximum exposure level that you can enter in the GUI so it won't crash the D405 camera an the GUI?

This may also be related to another issue which is having some of the supporting threads crash can crash the GUI. Will create another issue around this :)

TimKeesRensen, 2023-08-03:
I think it only crashes the D405 camera since the exposure setting is invalid - need to check value valid before sending to camera

TimKeesRensen, 2023-09-18:
Need to remove exposure controls from GUI

closed 2023-09-18 by TimKeesRensen