uc-vision/RovVision2 #59: Make automated scanning mode check to see if the ROV is in X and Y hold mode.

codedraughtsman, 2023-07-27:
I had several cases in our last field trip when I forgot to put the ROV into X and Y hold mode before starting an automated scan.
This causes the ROV to descend down the dropper line without moving to the correct distance to the line and moving side to side to track it.

Can we please implement one of these changes:

1. Make it so that the automated scan won't activate unless the ROV is in X and Y hold mode. It flashes a warning message to the user saying something like "please activate x and y hold mode"
2. Make it so that starting  the automated scan will also switch the ROV into X and Y hold mode automatically. Then when the scan is aborted, it reverts the X and Y hold mode to whatever it was before starting the scan.
3. 
I personally favour option 2

orig74, 2023-08-05:
I think in terms of implementation it will be better to set the XY hold mode when you start auto scan and leave it that way after the scan (option 3 :). 

If there are currents the ROV can drift in to the ropes at the end of the mission.

BTW,
In the operational GUI I am not giving the operator any choice between XY hold depth hold or ATT hold. There is only HOLD mode which tic all of the above. 
but I agree that XY hold should be activated also in engineering GUI

orig74, 2023-08-19:
Done

closed 2023-08-19 by orig74