uc-vision/RovVision2 #63: Detecting bottom of the dropper lines during automated scan not working

codedraughtsman, 2023-07-27:
In our last field trip (19th July 2023), detecting the bottom of the dropper lines during the automated scan did not work. 
The ROV would continue travelling down until it lost the dropper line and starting driving forward under the dropper lines.

orig74, 2023-07-30:
This is a critical bug since it can cause entanglement. Please try to provide time of event.

codedraughtsman, 2023-07-31:
@orig74  agreed that this is a high priority issue. 

We do currently have a work a round where we set the bottom of the scan depth to a meter or two below the bottom of the dropper line.

So this isn't actually impacting operations in the field yet.

As for event times, I don't have any logged times for it. My best guess for finding it is to look through the log files to find when we set the scan depth to a deeper depth than usual.
Most of the time we had set the maximum scan depth to 12(?) meters, which was before the end of the line at 14(?) meters.

Can you find the events?

orig74, 2023-08-08:
found event of not stopping at the end of the rope on video 230719-104817 approximately 120 sec from the beginning. the rope detector detects that this is not a valid state and not trying to maintain range. The problem is that it continues to go down. I fixed it and tested that it works on the recorded video an in simulation. it is fixed for now in the surveyour_dev branch. @codedraughtsman please run more tests in sim to see if I missed something.

closed 2023-08-08 by orig74

codedraughtsman, 2023-08-10:
@orig74 Will run some more tests in the sim sometime next week.

Will test it on our September field trip :)