uc-vision/RovVision2 #64: Automated scanning mode sometimes switches dropper lines

codedraughtsman, 2023-07-27:
When the dropper lines are close together, the automated scanning mode sometimes switches to an adjacent line.

The most common case is outlined below:

- Rov is descending down dropper line.
- A second dropper line is visible in the D405 camera and is around the same distance from the camera.
- The second line gets closer to the camera or the ROV drifts slightly in to the centre  between them. Sometimes the tracked line has part of it drift out of the side of the camera's FOV. 
- The ROV changes from the currently tracked line to the second line.
- User then has use the gamepad to shift the ROV back to the original line. (This doesn't interfere with the automated scan).

orig74, 2023-07-30:
I would classify it as non critical, and see if it affects our estimations in the future.

codedraughtsman, 2023-07-31:
@orig74 Agreed, this is just a minor issue right now and we have bigger fish to fry :)

closed 2023-09-18 by TimKeesRensen