uc-vision/RovVision2 #65: Event of interest button

codedraughtsman, 2023-07-27:
When we are out on a field trip, we are collecting huge amounts of data.

Trawling through all of that footage and data is not feasibly possible.

Can we have a button in the GUI (and on the controller) that marks that an "Event of interest" has occurred  and stores it along with the rest of the data?

Then later we can go at analyse the marked Events of interest. We could even look into creating a viewer that automatically plays/extracts all the events of interest for us.

Typical Events can include:

- ROV switching between tracked dropper lines during an automated scan.
- ROV trying to go between lines
- One of the ROV/GUI processes crashing

orig74, 2023-07-30:
For the first 2 points Good Idea! will add that. For the last one if the GUI has crushed we cant use a button :) but you are right any crash should be logged and easy to find.

orig74, 2023-07-30:
added LOG button.

closed 2023-07-30 by orig74