uc-vision/RovVision2 #67: Auto adjust and set exposure setting for cameras.

codedraughtsman, 2023-07-27:
Can we add an auto-exposure mode for the cameras (D405 and Vision)?

Simply put, we create a histogram of all the pixel values in the image and adjust the exposure settings until histogram covers the entire value range.

This should be a reasonably solved problem, given how many cameras do this. We may have issues that the red spectrum is underexposed more than the blue spectrum. 

It would also be nice for the histogram to be displayed to the user.

orig74, 2023-07-30:
D405 has auto exposure mode like most cameras for the entire image. The problem with auto exposure is that it is not sensitive to the areas of interest (the lines). we can detect the lines and try to limit the exposure algo to the lines maybe. For now I am suggesting to wait for the strobe system to work and rethink this after we collect some data. Keeping this issue open for further considerations.

codedraughtsman, 2023-07-31:
@orig74 I agree that we should hold this issue until the strobe system is further finalised. 
We could start working this for the vision cameras, but it makes more sense to do both the vision and the D405 at the same time.

TimKeesRensen, 2023-08-03:
I think we should auto-adjust the D405 exposure and the Allied vision gain based on depth, I will make these changes.

orig74, 2023-08-04:
The exposure will be highly depends on the strobe system. if the strobe system works short exposure will be preferred no matter the depth. I am also hoping the the strobes will have positive effect on better identifying the front line since its going to be much better lit.

TimKeesRensen, 2023-09-14:
With the strobe system working we shouldn't need to change exposure which is set to 200us-400us on both cameras but the gain will need to be adjusted.

TimKeesRensen, 2023-09-18:
manual camera controls with auto modes - gain, strobe / exposure duration