uc-vision/RovVision2 #68: Don't detect lines on the other side of the backbone.

codedraughtsman, 2023-07-27:
On our last field trip, we had issues that the automated scan would lock on to the dropper lines on the far side of the backbone.

To fix this we lowered the maximum detection range of the D405. 

However this did seem to cause issues when the automated scan was enabled when the ROV was far away from the dropper line we were trying to scan. 

Can we look at adding two maximum detection ranges? One that is used when approaching the line for the first time and another for when it has started scanning the lines.

Perhaps we need a new state, approaching_line, that moves the ROV into position next to the dropper line at the start of the scan.

orig74, 2023-07-30:
I think approaching_line state or something similar is a good idea. I will look into this.

orig74, 2023-08-19:
After giving it some thoughts, I don't like the idea (for now) of approaching the lines autonomously. It's unnecessary risk. maybe we can come up with an operating procedure of manually approaching the rope, verify that visibility is good and the lock on the rope is valid (white line mark in the upper top of the image )  and only then start the scan.  I can automate it in the future to not letting the scan start and give a message in case conditions are not good for scanning.

codedraughtsman, 2023-08-21:
@orig74 Manually approaching the rope then enabling the scan worked on the last field trip. There are more important tasks to do right now than this one. 

Agree that we can revisit this once the process/software becomes more mature.

TimKeesRensen, 2023-09-18:
Need more data, condition based tracking?