uc-vision/RovVision2 #70: GUI not setting D405 exposure on startup.

codedraughtsman, 2023-07-27:
When starting up the GUI, the exposure setting for the D405 is not been used by the ROV/D405.

Steps:
- Set the D405 exposure setting in the GUI to 4.
- Restart the GUI
- The video coming back from the D405 shows footage which has the incorrect exposure setting.
- Changing  the exposure value in the GUI correctly changes the GUI.

My guess is that the GUI isn't sending the exposure setting to the D405 when it loads the settings from last session.

orig74, 2023-07-30:
Hi this is definitely a bug. We currently don't have a process of sending current saved data on GUI side to the ROV on startup. I will fix that soon.

orig74, 2023-08-05:
fixed

closed 2023-08-05 by orig74