uc-vision/RovVision2 #71: Control event of interest button (Log) from controller

codedraughtsman, 2023-07-31:
https://github.com/uc-vision/RovVision2/issues/65 is unclear if we can activate the log button using the Gamepad.

Can you add this? Usually, the operator doesn't have enough time to click buttons on the GUI using the mouse.

orig74, 2023-08-04:
Done

codedraughtsman, 2023-08-10:
@orig74 What is the key combo to create a Log event?

orig74, 2023-08-10:
It's not a key combo just a button named LOG

codedraughtsman, 2023-08-18:
@orig74 Apologises, I see that the request was ambiguous.

It should be:
Can we have a way to create a Log event by pressing some key combo on the controller?

codedraughtsman, 2023-08-18:
Have updated the title and description of this ticket, to be less ambiguous.

orig74, 2023-08-19:
No need for special key combo. every time you presses a key (in recording state) in the game pad it is recorded so any unused combination I can detect and review in the log file. If possible use the the LOG button in the GUI since it will also include the text from the text box.

closed 2023-08-19 by orig74

codedraughtsman, 2023-08-21:
@orig74  I understand. 

Not sure how I feel about having any key combo be a stand in for a logged event. Would prefer a dedicated key combo, but this isn't important right now. Will invent a unused key combo for logging events.