uc-vision/RovVision2 #79: GUI resizing/screen resolution

codedraughtsman, 2023-08-28:
When testing the ROV basestation/ control GUI  on our older backup laptop.

we found that the screen resolution was different to the newer laptop and the right hand part of the GUI was off of the display screen.

backup laptop resolution 1600x900
Original laptop resolution  ?

We need to fix this going forward. 
Right now, this isn't a problem. It will only matter if we need to start using the back up laptop.

orig74, 2023-08-28:
I am suggesting closing this bug. This is a >6 years old laptop. In the past I supported low screen resolution and it requires all sorts of rearrangements to the GUI layout. I thing that Full HD res is pretty standard today so getting a new backup laptop makes much more sense.

codedraughtsman, 2023-09-10:
@TimKeesRensen Can we get a new backup laptop? 
I'm happy to close this, as long as we have a functioning backup laptop/GUI.

closed 2023-09-18 by TimKeesRensen