uc-vision/RovVision2 #82: Brownout issues

codedraughtsman, 2023-09-12:
We currently just run the ROV until it browns out and needs to be retrieved. 

Usually the brownout occurs when we use all the thrusters at the same time and we draw a lot of current from the battery.

Johann and Timothy will hook up an oscilloscope to the ROV in the lab and define the characteristics and path of the brownout.
It may be able to be solved by adding a capacitor or rerouting the main current draw away from the control electronics. 

For the software side, we can limit the maximum thrust limit based off of the battery voltage and current draw. 
Also, we can implement a low battery alarm/warning for the ROV.

Ideally we can also make it so that all the ROV systems restart and connect back up to the base station.

TimKeesRensen, 2023-09-18:
Check pkl is flashing during recording, add battery low warning, make sure panda restarts correctly

codedraughtsman, 2023-12-14:
We now have issue #116 which shows a battery percentage on the GUI

jozefeijoa, 2023-12-18:
Can we move this issue over to ROVMisc repository as it's hardware-related?

@codedraughtsman has recommended that all hardware-related issues be in the ROVMisc directory since the RovVision2 repository is software, strictly speaking

codedraughtsman, 2023-12-19:
@jozefeijoa Sounds good. 

feel free to move issues around as needed. Instructions on how to do so are here :) https://docs.github.com/en/issues/tracking-your-work-with-issues/transferring-an-issue-to-another-repository

codedraughtsman, 2023-12-19:
@TimKeesRensen Going to solve with thruster clipping in ESP32 gate.

TimKeesRensen, 2024-01-16:
Needs extensive field testing & tuning