uc-vision/RovVision2 #135: Driver reliability and autonomy enhancement

jozefeijoa, 2023-12-19:
Increase the reliability of the sensor drivers so that as many possible errors on driver boot are caught, and make the drivers as self-correcting as possible, with only the fatal, unrecoverable driver errors reported to the GUI

codedraughtsman, 2023-12-19:
Seems similar to #119

jozefeijoa, 2023-12-19:
> Seems similar to #119

Thanks Timothy, I have marked the other as duplicate and cleaned it up.

jozefeijoa, 2024-01-10:
Latest commit on branch drv-improv has implemented the changes and fixes in #132 .

Won't fix #136 as this was likely erroneously reported.

jozefeijoa, 2024-01-15:
The frequent DVL script crashing on boot has been fixed by upgrading the DVL software.

According to @TimKeesRensen , the Mako driver reliability has been improved (on the surveyour branch) from changes he has made.

The only task that remains, and a new one, is to purchase a new D405 camera because the D405 currently in the secondary ROV likely has hardware damage and will lead to unreliable and sporadic faults in operation (sometimes it stops responding and has a hardware crash, necessitating a power cycle).

jozefeijoa, 2024-01-15:
Updating the DVL software was fixed the DVL script startup reliability issues.

According to @TimKeesRensen , the Mako driver's reliability has been improved from his code improvements on surveyour branch.

The only remaining task, and a new one, is to buy a new D405 and replace the existing D405 on the ROV since the existing D405 likely has hardware damage, which means that it is unreliable. It's current symptoms are that it hardware crashes sometimes, necessitating a power cycle to fix the issue.